DnG-Crafts/K2-RFID #1: OTA Updates

lowclearance, 2025-02-26:
This is more of a request & suggestion for a future update.  

Can you include the [ArduinoOTA](https://github.com/JAndrassy/ArduinoOTA) library and functionality so that the board can be updated over Wifi without moving it?

My setup is in my garage but my Arduino IDE computer is in another part of my house.  I know this may seem lazy buy I made the setup semi-permanent so moving it is difficult and moving the PC is as well.  I could add the code but if you update your code or the database changes then I will need to add the code each time before updating.

I really appreciate the code and project and use it every time I get new filament.  All of my current non-RFID filament have tags now as well.

Thank you!

DnG-Crafts, 2025-02-27:
added a function to update the firmware via the webpage.
use the arduino ide to export compiled binary(.bin) and upload it through the web updater.

closed 2025-02-27 by DnG-Crafts

lowclearance, 2025-02-28:
You are awesome! 😎 Thank you!

DnG-Crafts, 2025-02-28:
> You are awesome! 😎 Thank you!
> 
> Quick question: Do all the .bin files need to uploaded or just the Spool_ID.ino.bin?

You should only have one bin exported  spool_id.ino around 1mb in size if its the esp32 sketch