DnG-Crafts/K2-RFID #4: no other brands than Crealty and Generic

Michil783, 2025-04-15:
Hello,
really great app (both Android as well as Arduino). I have an issue with a self modified database file. I added an own filament brand with all necessary data in it. The printer accepted the file and is showing my own brand. Arduino as well as Android is readinf the file from the printer but still showing only the two basic brands.

Any suggestion what's wrong?

I added the db file I'm using.

[material_database.json](https://github.com/user-attachments/files/19763310/material_database.json)

DnG-Crafts, 2025-04-15:
i have updated the android app to show the vendors correctly and tested your material_database file to confirm it works.
i will update the arduino code when i get some time to work on it.

closed 2025-04-15 by DnG-Crafts

Michil783, 2025-04-16:
I have to change my DB again due to identification issues with the printer. Now I defined 20001 in „id“ as „Das Filament PLA“, 20002 as „Das Filament PETG“ and 30002 as „FilamentWorld PETG“
If you like I could provide my DB which is now also working on my printer successfully.