DnG-Crafts/K2-RFID #5: WIFI

ginwer7619, 2025-04-20:
Hello,
I installed everthing like described but I am unable to connect to the ESP32 via WIFI. 
The rest works fine, connecting via the Acces point and writing tags works without any issues. 
I put the Wifi Name and the wifi code in the respective fields in the Arduino but the ESP 32 does not connect and does not appear on the router. 
Where can I find a solution for that. 
Thank you

DnG-Crafts, 2025-04-20:
are you putting the wifi info into the code or are you setting it through the web ui?

DnG-Crafts, 2025-04-21:
not sure why it is connecting.
maybe try some example wifi arduino code from the esp32 library to make sure your router is allowing the esp to connect.

it might have something to do with the [auth mode settings](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/network/esp_wifi.html#_CPPv416wifi_auth_mode_t)

DnG-Crafts, 2025-04-21:
not sure why it is connecting.
maybe try some example wifi arduino code from the esp32 library to make sure your router is allowing the esp to connect.

it might have something to do with the [auth mode settings](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/network/esp_wifi.html#_CPPv416wifi_auth_mode_t)

https://github.com/espressif/esp-idf/blob/b4268c874a4cf8fcf7c0c4153cffb76ad2ddda4e/components/esp_wifi/include/esp_wifi_types.h#L60-L75

```
WiFi.setMinSecurity(WIFI_AUTH_WPA2_WPA3_PSK); 
WiFi.begin(WIFI_SSID.c_str(), WIFI_PASS.c_str());
```

closed 2025-05-20 by DnG-Crafts