DnG-Crafts/K2-RFID #6: custom filament settings?

EQNish, 2025-05-13:
Is there a way to add/change the settings embedded?
name or anything that is tied to the filament profiles

calvarez, 2025-05-13:
As in changing or adding other brands to the printer via RFID? I’ve been trying to do that but can’t figure out any way to do it.

DnG-Crafts, 2025-05-13:
> As in changing or adding other brands to the printer via RFID? I’ve been trying to do that but can’t figure out any way to do it.

you cant add data via rfid, the rfid just identifies filaments in a database on the printer.

if you wish to change things in the filament database you need to modify the database on the printer which requires you to root the printer and use tools to modify the files on the printer

the database for settings is this file
https://github.com/DnG-Crafts/K2-RFID/blob/main/docs/material_database.json

this is the directory for the filament data `/mnt/UDISK/creality/userdata/box/`

closed 2025-05-13 by DnG-Crafts

EQNish, 2025-05-13:
I noticed while playing that esun has a rfid name on the printer...not sure if that helps!

calvarez, 2025-05-13:
I have been unable to get this application to write the eSun info, however. That’s something I just don’t understand. How to expand the ability to write other information to the tags themselves. Using the button to get data from the printer doesn’t seem to do anything.

Maybe the author could add a brief summary on where to modify the data that will be written to the tags? I’ve tried changing various JSON files for the Arduino. But I’m not a dev, I’m a network guy. I just make the pipes.