DnG-Crafts/K2-RFID #7: No longer able to update filaments

haloweenhamster, 2025-05-17:
Hi previously I was able to update the filament list, it would only support creality or generic for brand on the app but as it went by the ID no. The correct brand was shown on the printer / slicer
As can be seen here
[image]

[image]

The app has updated and now only allows updating via GitHub and not the local printer

Is the feature of updating via printer likely to return or is there a way to get an older version 

Thanks 

PS brilliant app, thanks for your time and effort that you've put in to bring this to people

DnG-Crafts, 2025-05-17:
creality has changed things so updating from the printer does not get the correct database file.
they are updating it in the background from the printer directly to the creality servers and not via firmware updates and the file they are using is not exposed to the frontend folder so it cannot be accessed via the app using http.

they appear to also have different database files with different filaments for the different printers as in the Hi has a different set of filaments to the K2.

so i will have to add printer type selection to the app.

updating from the printer is really only useful for people that create custom database lists but i suspect creality will overwrite those in the background.
i am still trying to find the url they are pulling the updated database files from so the database file can be downloaded directly from creality.

i will add the printer update method back as a selectable option for people that tinker with the files.

DnG-Crafts, 2025-05-20:
i have added back the option to pull the database from the printer and have also added support for the k2, k1 and hi printers.
google will roll the update out over the next few hours.

closed 2025-05-20 by DnG-Crafts