DnG-Crafts/K2-RFID #10: Updating printer DB crashed printer, now doesn't update

calvarez, 2025-06-15:
For my first attempt at creating a new filament, I made one and changed just a few of the details to match, then uploaded to the printer.  I had all the defaults for this app (reboot printer on).  Once the printer rebooted, it gave a system error and did the thing where it drops the bed and turns off all the lights.  All you can do is power off/on, which restored the printer to normal operation.

After that, the filament doesn't show up in the list on the printer to be selectable.  Should it be visible, or is it only going to be usable with the NFC tags?  The software says it has updated the database.  I've also tried having it not reboot the printer and then I do it manually, but the new filament still doesn't show up in the screen on the printer.  The app always says that it has updated the printer and is rebooting (when I have the reboot option checked).

scbowes, 2025-06-15:
You have not said what system you are using Android , Arduino or Windows. If your using windows what version. 

I am not experiencing  any issues  with Android works perfect

calvarez, 2025-06-15:
A fair point, I didn't realize the Android version could update the printer database.  

I installed the Windows version 2 from the compiled download here last night.

DnG-Crafts, 2025-06-15:
if you dont have `Prevent DB updates` ticked when the printer reboots creality will check try to update the database in the background and it removes your changes.
there is no way in the printer menu to turn this off so the app has the option to prevent the updates.

i just modified the code so it will pull the current version of the database off the printer when you upload the database if you do not have prevent db updates ticked but this is only a short term option as creality will push an update.

closed 2025-06-18 by DnG-Crafts