DnG-Crafts/K2-RFID #14: Error message:  "NFC reader failed"

calvarez, 2025-06-18:
Using V5 pre-compiled, downloaded this morning on a fairly new Windows 11 machine with all updates.  The ACR drivers were the latest available today.  The ACR122u reader is from the ACS store on Amazon, brand new, and appears to be legit and working.  I've rebooted and powered off multiple times and tried several USB ports as well as a powered hub.  If I swipe a blank tag the reader beeps and goes green, but of course the tag is left blank.  The tags are the same ones I use with your Arduino/Pi version and work great.

[image]

DnG-Crafts, 2025-06-18:
something else to try is this copy, it does not embed the resources into the app.
embedding them might be the issue.

[CFS-RFID_NE.zip](https://github.com/user-attachments/files/20803672/CFS-RFID_NE.zip)

calvarez, 2025-06-18:
Ugh, sorry, problem solved, it was MS RDC.  With your new build, I got this:  The smart card resource manager is not running.  A little digging led me to believe it was redirecting the card reader to remote desktop, which I was using since this machine is in the printer room elsewhere.  Even though I have smart card redirect disabled.  Leaving the error check/display in future versions would be a good idea I think.

Making the RDC connection AFTER starting your program does work.  It will generate an error on read, but still read.  It will write just fine with no errors.  You can't start the program via RDC however (works other than the tag reader).